Gissio/radpro #209: Glitch in measurement readings (2.1)

denka977, 2025-02-15:
at elevated radiation levels the dot disappears for a couple of seconds, then returns (firmware 2.1)

[image]

STGitUser, 2025-02-15:
The decimal also dissappears temporarily if the CPM drops below 10.

STGitUser, 2025-02-15:
[image]
[image]
[image]
[image]
[image]

deinotoxazumab, 2025-02-28:
I have a theory, but it's a long shot: What if Open Sans is just ever so slightly too wide for the device to fully display compared to Roboto in previous releases?

I was toying around with the fonts in the whole Repo to insert custom fonts in. It is a success and along the way, I found similar issues as this issue, i.e. the decimal point may occasionally disappear. If the fonts are even wider, it won't be just the decimal points that disappear, but also certain numbers. I used Font Forge to manipulate the font glyphs width to be ever so slightly narrower, and it alleviated the issue; the decimal point stays visible now. Now I don't know what exactly happens when the font's glyph width on the device is a little too wide on the technical level, especially since the font on 2.1 has been replaced from Roboto to Open Sans.

If anyone wants to test my theory, feel free to. Try using fonts that are narrower than the stock one preferably Condensed (narrower) fonts as a positive control. I have, but I'm afraid it might feed to my confirmation bias so I need anyone else giving it a try. All you need to do is:

- Replace the fonts with narrower fonts of your choosing
- Alternatively, you can manipulate existing fonts in the Repo using Font Forge, but it's a bit too much work
- Run supplied fontcov program on them
- Build the firmware with PIO on VScode then sign it
- Flash it on the device. GC-01 is the easiest to flash

Using Condensed fonts may look ridiculous by leaving tons of horizontal screen real estate on the table, but it's to make sure whether the decimal point stays visible all the time. From there you may iteratively widen the fonts with Font Forge.

Let's hope this Issue is purely cosmetic.

Gissio, 2025-02-28:
The issue is in mcu-renderer. I plan to fix this once I have time.

Gissio, 2025-03-11:
Update: the culprit is a ">" comparison, that should be a ">=". Fixed in the upcoming beta.

Gissio, 2025-03-11:
Can you try https://github.com/Gissio/radpro/actions/runs/13797030926/artifacts/2732897104 and see if the issue is fixed?

dc1rdb, 2025-03-12:
The dot is back with 2.1.1 beta 1 on my Geehy GC-01.
Many thanks for the fix!