Gissio/radpro #212: FNIRSI GC-02 - anyone has these new Geiger counters?

mryndzionek, 2025-02-18:
Here is an example [link](https://allegro.pl/oferta/fnirsi-gc-02-licznik-geiger-detektor-promieniowania-jadrowego-17003829732).

Gissio, 2025-02-18:
It is my understanding the case of the GC-02 does not accommodate J321 or similar tubes. Can you confirm?

mryndzionek, 2025-02-19:
> It is my understanding the case of the GC-02 does not accommodate J321 or similar tubes. Can anybody confirm?

I don't have it, just asking if someone here has it, but looking at YT videos and reviews on Amazon, yes, it is too small for J321...

IWasBorg, 2025-02-21:
It appears to be a smaller J613 previously used in certain GC-01 batches after J321 and before M4011.  MCU appears to be a WCH CH32F203 but below's photo is blurry

Source: [https://www.elektormagazine.com/review/fnirsi-gc-02-nuclear-radiation-detector-review](url)