Gissio/radpro #213: Fallo en modo data mode

Vteriaza, 2025-02-20:
Hola, resulta que al poner en el menu data mode en on, deja de leer el voltaje de la bateria y aparece 00000v y el simbolo de la bateria aparece vacio.
al ponerlo en off , vuelven a aparecer la lectura de la bateria y su voltaje. Gracias.

Gissio, 2025-02-21:
Can somebody else confirm this? A short video would be helpful.

Vteriaza, 2025-02-21:
perdona pero no se como subir el video, no me deja.

Gissio, 2025-02-21:
You didn’t mention this happens when turning the device off and then on. Thanks for the report!

TakissX, 2025-03-11:
The same thing happens to me. As it does not show the correct voltage when DATA mode is not activated

Gissio, 2025-03-13:
Can you try https://github.com/Gissio/radpro/actions/runs/13840899966/artifacts/2747445830 ?

It will be noisier than in the regular Rad Pro, but the voltage problem should be fixed.

Gissio, 2025-03-26:
Should be fixed in 2.1.1.

closed 2025-03-26 by Gissio