Gissio/radpro #215: FS600 cell voltage drops to 0.808v after being powered on for a while

oldmanav, 2025-02-21:
Hi, I have installed Radpro V2.1 on my Bosean FS600 via STLink. I have noticed that after a while, the battery indicator will show the battery as empty and the statistics show the battery voltage has dropped to 0.808v. Switching the device off and on again returns the battery voltage in the statistics section to the actual voltage (1.2xxV). 

I have confirmed this with a multimeter - when 0.808v is displayed, the individual battery voltages are actually 1.2v.

Gissio, 2025-02-23:
Thanks for the report.

Gissio, 2025-03-13:
Can you try https://github.com/Gissio/radpro/actions/runs/13840899966/artifacts/2747445830 ?

In this test release, voltage filtering is disabled, thus the Statistics view should show the instantaneous voltage value (which is much noisier than the regular voltage view as it is not filtered).

Gissio, 2025-03-13:
Can you try https://github.com/Gissio/radpro/actions/runs/13840899966/artifacts/2747445830 ?

In this test release, voltage filtering is disabled, with the Statistics view showing the instantaneous voltage (it will be noisier as it is not filtered).

It would be great if you can follow the voltage and see when/how the voltage starts showing incorrect values.

oldmanav, 2025-03-15:
Hi,
I have flashed this firmware and I will monitor the behavior of the device. I have put a slightly lower capacity set of cells in the device., so they should discharge slightly quicker. Funnily enough, I had left the device on with the previous firmware at a later date and the issue did not reoccur.

oldmanav, 2025-03-19:
No issues to report so far, however this issue may be intermittent

Gissio, 2025-03-26:
Should be fixed in 2.1.1.

Gissio, 2025-04-01:
It seems that Rad Pro’s measured battery voltage is 2/3 of the actual voltage. Can somebody confirm this?

oldmanav, 2025-04-01:
Hi Gissio, 

The measured voltage in the statistics page running the 2.1.1beta2 firmware is 1.25v. My multi meter measures 3.40v across the battery terminals.

I always assumed the value was 1/3 of the actual voltage, i.e. the value of a single cell as it is a 3 cell string. I use nimh batteries with my device.

Gissio, 2025-04-01:
Hello oldmanav:

Thanks for the report! That’s the way it’s expected to work.

Could you please check with the 2.1.1 release? @dg0mg is reporting 2/3 of the expected voltage, as you did when you raised this issue.

Gissio, 2025-04-01:
So there is no bug at all.

The statistics view shows "Device voltage" quite simply because the label "Device battery voltage" does not fit.

What you call + is actually a low-pixel lighting symbol.

closed 2025-04-01 by Gissio

DG0MG, 2025-04-01:
I dont understand?
Why do you write it isnt a bug, when the Battery has 3 Volts and you show a "Device Voltage" of 1.025 Volts?

Gissio, 2025-04-01:
Let me cite from my previous answer "The statistics view shows 'Device voltage' quite simply because the label 'Device battery voltage' does not fit". So I am implying that the value is per cell.

Gissio, 2025-04-01:
The value is per cell. Do you think it would make more sense to display the value across all batteries?

DG0MG, 2025-04-01:
OF COURSE!
Nobody would think that the voltage of a single cell is specified :) A lead-acid car battery also has 10.5 to 12 volts, even though the individual cell is somewhere around 2 volts. Especially since the individual cell voltage is technically impossible to measure. A "normal user" ^^ often puts cells with different charge levels into a flashlight, radio, or something similar. The only thing that can be reliably measured is the total voltage.
But that's MY opinion. I never thought that this was the voltage of a single cell, but @oldmanav did. I'd like to hear other people's opinions.