Gissio/radpro #216: Bosean fs-5000 wrong voltage

notwired, 2025-02-23:
Voltage shown in statistics shows an impossible 4.862 V but measured with voltmeter is around 4 V.
After reset, voltage shown in statistics is near measured voltage but after 2 - 3 hours its showing 4.862 V again.
Tested only on latest firmware 2.1

[image]

Gissio, 2025-02-23:
Seems to be the same issue as #215, but for a different device. Thanks for the report.

Gissio, 2025-03-13:
Can you try https://github.com/Gissio/radpro/actions/runs/13840899966/artifacts/2747445830 ?

In this test release, voltage filtering is disabled, thus the Statistics view should show the instantaneous voltage value (which is much noisier than the regular voltage view as it is not filtered).

It would be great if you can follow the voltage and see when/how the voltage starts showing incorrect values.

Gissio, 2025-03-26:
Should be fixed in 2.1.1.

closed 2025-03-26 by Gissio