Gissio/radpro #217: FNIRSI GC-01: flashtool fails silently on very broken devices

shpank, 2025-02-24:
Hi,
First of all: My heartfelt thanks to all developers of radpro. This Firmware makes these cheap devices so much better.
I was kind of an "early adopter" meaning i tried loading some beta firmware for the GC-01 before reading the instructions and warnings about its unstable nature. So naturally, i bricked my geiger counter!

After waiting for a more mature version of radpro, I finally found out about the alternative install mode via the ST-Link debugger, so I had to try it today. But I encountered an issue during flashing:

The flashtool.sh failed silently after halting the CPU with "Backing up old firmware image...". Further examination revealed that it exited with code 3 which meant that it was unable to backup any existing firmware. I patched the flashtool.sh to skip the firmware download part and tried again. This time, I was able to successfully flash version 2.1 onto my geiger counter.

My question: Could the error be stated in a more obvious way or could you provide a way to do an install/update without trying to run a (in some cases pointless) firmware backup?

Gissio, 2025-02-24:
Thanks for the report!

In fact, there is an error in the install and update scripts: they attempt to download the original firmware image, but the FNIRSI GC-01 comes factory-locked, meaning that such a download always fails.

Thanks for pointing this out.

Gissio, 2025-03-13:
Could you try this release and see if the script now works?

[radpro-flashtool-2.1.1beta2.zip](https://github.com/user-attachments/files/19234730/radpro-flashtool-2.1.1beta2.zip)

Gissio, 2025-03-26:
Should be fixed in 2.1.1.

closed 2025-03-26 by Gissio