Gissio/radpro #222: Bosean FS-5000: Battery indicator might be faulty

Gissio, 2025-03-09:
### Discussed in https://github.com/Gissio/radpro/discussions/221

<div type='discussions-op-text'>

<sup>Originally posted by **TheAsgard** March  8, 2025</sup>
I have had this installed since launch and it been perfect. Battery life is now unbelievable. Not sure what you have done but it’s still showing full after 2 weeks+ powered up.

What did you do to improve battery life so much compared to the previous version?

Thanks for all your efforts. </div>

starchaser01, 2025-03-09:
After a while, the value of the battery voltage starts to creep up, and it can't be reversed. It's interesting to note that the time until this happens seems to depend on the strength of the radioactive radiation. While it normally takes hours, it can happen after just a few minutes near a radioactive source.

starchaser01, 2025-03-13:
I think I tried something like this this myself and took the battery voltage directly without the reference value.

With the old firmware, the battery voltage was previously at 3.7V immediately after switching on. Now the voltage is at about 4V and stable. I then held the device to a test source for a long time to provoke the error, without any result.

I don't know what's going on. Maybe it's a fault in the hardware design that somehow changes the reference voltage due to the radiation.

Gissio, 2025-03-13:
That's actually great news. Could you measure the voltage at the battery, and confirm it matches what the statistics view shows?

starchaser01, 2025-03-14:
I've taken a closer look. The voltage displayed on the FS-5000 was 4.007V and it fluctuates a little in the millivolt range as expected. When measuring the battery directly with a multimeter, the voltage was 3.66V.

Apart from the discrepancy in the voltage, there is another problem. After some time, the displayed voltage is no longer updated. Only a restart will get it working again.

So basically the original problem still persists, just in a different way.

Gissio, 2025-03-14:
Thanks for your effort.

Could you measure the time it takes for the voltage to stop updating?

starchaser01, 2025-03-14:
Did a couple of tests. The time it takes for the voltage to stop updating varies. 4 Minutes, 9 Minutes, 19 Minutes. No fixed pattern. Seems to be influenced by cpm somehow.

Gissio, 2025-03-14:
Very useful.

When the voltage stops updating, does it stabilize at approximately 4 volts, or does it settle at a different value?

Gissio, 2025-03-14:
Your report suggests that the ADC data register stops getting updated.

Now I have something to work on. Thanks.

Gissio, 2025-03-15:
Can you try this test: https://github.com/Gissio/radpro/actions/runs/13874297323/artifacts/2757892958 ?

starchaser01, 2025-03-15:
I tried it and after about 3 minutes it unfortunately hung again.

What I noticed yesterday and today is that the pulse clicks are sometimes not reproduced correctly at higher cpm, so that occasionally there are getting replaced by short "beeps" of different lengths.

starchaser01, 2025-03-20:
I took another look at it today and found a small workaround. Increasing the wait time after the one shot ADC conversion seems to have fixed the problem. A better solution would be to wait for the EOC flag that is set after the conversion.
 
It's probably also not necessary to check the charging status so frequently. Checking every 10 seconds, or an even longer interval, should be more than sufficient.

Gissio, 2025-03-23:
Thanks for your feedback!

I'm intentionally not using EOC so in case the ADC fails, the counter can still function. So I'm relying on the conversion time specified in the datasheet to ensure completion.

I'm also sampling the battery voltage once per second. This is because the power consumption of other devices tends to fluctuate significantly due to the switching power supply. I'm not certain if the FS-5000 behaves similarly, but keeping a common codebase across devices simplifies development.

I've reviewed the code, but unfortunately, I haven't found any obvious issues. The conversion should take 0.1 ms, which is well within the 1 ms delay I'm using. Yet it seems the ADC's OVERRUN flag is being triggered.

In response, I’ve increased the ADC clock frequency in beta4:

https://github.com/Gissio/radpro/actions/runs/14016199577

Could you please try again and let me know what happens?

starchaser01, 2025-03-23:
Thanks for the detailed explanation.

I tried Beta 4. Unfortunately, after a few minutes, the ADC also stops providing new values there.

By the way, I had increased the waiting time to 10ms in my test, in case this information helps. With this setting, it ran smoothly for days, even at high CPM.

Gissio, 2025-03-24:
Let's try one more time, this time with EOC:

https://github.com/Gissio/radpro/actions/runs/14044292193

starchaser01, 2025-03-24:
The test is running, and after about 60 minutes it still looks good, but I'll give it at least 12 hours at high CPM load to see if everything is ok.

By the way, I see the old button logic is back. I have to say that I prefer it. It fits very well with the print on the buttons.

Gissio, 2025-03-25:
Great news!

DMA wouldn’t make sense here.

starchaser01, 2025-03-25:
No problems after 12 hours.

(The voltage is of course still displayed as too high. That was fine in 2.1, at least as long as it worked).

Gissio, 2025-03-26:
So I guess this is fixed!

closed 2025-03-26 by Gissio