Gissio/radpro #223: graphic readability on fnirsi GC-01

Turbo-poulet, 2025-03-12:
Hello, and first of all, thank you for your tremendous work, which you have kindly made available to us, it's truly exceptional.

I would like to make a request regarding Radpro's FNIRSI GC-01. 

[image]

When we put the graph up, there is no height scale indicated on the side.
Sometimes there's even no grid, just columns.
Adding the orders of magnitude as on the original firmware would make the graph easier to read.

Thank you very much for what you're doing once again.

Gissio, 2025-03-12:
Let me cite from the manual: “The plot uses a [logarithmic scale](https://en.wikipedia.org/wiki/Logarithmic_scale), where each division represents a tenfold increase in radiation level. Alert zones at 1 µSv/h and 10 µSv/h are visually indicated.”

In your image, the topmost line represents 1 µSv/h, the bottom line, 0.1 µSv/h. The scale is automatically adapted to the range of the data.

When measuring radiation, usually the logarithmic value matters most. That’s why there is no sub-decade precision in the plot.

Turbo-poulet, 2025-03-12:
Okay, however in this example the peak is at 1µSv, but on the graph you would think it went up to 5µSv given that it is between 0.1µSv and 10µSv.
For me, it's a bit unclear, and a scale on the side would make the graph more understandable, at least for a beginner like me.
In any case, thank you very much for your reply. I wish you a good evening (it's dark here in France).

[image]

Gissio, 2025-03-12:
The plot is logarithmic, that's why the manual says that "each division represents a tenfold increase in radiation level"!

Turbo-poulet, 2025-03-13:
Okay, I finally understood, I didn't know about the logarithmic scale, it's clearer now, indeed it allows you to have an idea if ever there should be big variations compared to a linear scale as I thought there was.