Gissio/radpro #233: switching hv configuration disabled led joy it jt-rad01

Androidonator, 2025-04-01:
After flashing the firmware the led and sound was working fine but after changing the configuration to energy saving it disabled led and sound and vibration.
Changing the preset back to factory didn't help. 
I had to reflash the firmware and then it worked again.
I really have no idea what could be causing it maybe wrong configuration of the factory default preset.

Gissio, 2025-04-01:
Can somebody else confirm this?

Gissio, 2025-04-01:
Did you use the -install or the -update version?

Androidonator, 2025-04-02:
I will try to reproduce it on the weekend and get a video of it.

Gissio, 2025-04-12:
What is the state of the options in the Pulses menu before and after the problem occurs?

Is the Pulses menu “Threshold” option enabled?

Gissio, 2025-05-04:
Closing as there is no follow up.

closed 2025-05-04 by Gissio