Gissio/radpro #234: History problem v2.1.1

Gerard-Rdam, 2025-04-01:
I have discovered a problem in 2.1.1 the history is not recorded properly after a few hours it does not go further than an hour and this seems to record irregularly.
it stops after a while and if I hold a radiation source to it then it starts again for a while and then stops again.
JOY-IT JT-RAD01

[image]
[image]

Gissio, 2025-04-01:
Can you confirm your Geiger counter works correctly? What does the instantaneous view show during such a lapse?

Gerard-Rdam, 2025-04-02:
Instantaneous = 30seconds
Average = 5 minutes
Sensitivity = SMB-20
2.5Khz/19% for 400Volt
Everything seems to be working fine so far.
For clarity, the measurement stops on the horizontal axis and becomes active again when approached with a radiation source.
On the 10min scale in the photo is a recording of approximately 45 minutes

[image]

[image]

[image]

[image]

Gerard-Rdam, 2025-04-02:
With the Adaptive fast setting the History works fine.
the Average setting was Unlimited instead of 5min but was also visible on my photo.

Gerard-Rdam, 2025-04-02:
Setting with 1 minute also works well.
setting with 10 seconds see photo

[image]

[image]

Gissio, 2025-04-02:
You're using a 30-second instantaneous rate averaging interval, which calculates the rate based on the pulses detected within a 30-second sliding window. If no pulse occurs during a given 30-second period, the rate will register as zero. This explains what you're seeing.

Gerard-Rdam, 2025-04-02:
ok that is a logical explanation thanks for this.
i now understand better how it works i used the 1 minute setting in the previous version. so this is not a problem. maybe for a next version a message or block on the history if the setting is lower than a minute?
is it also possible in the next version to set the HV standard setting to 2.5Khz/19% so that the standard HV is no longer 700 volts in connection with the safety of the geiger tube?
and further thanks for everything :-)

Gissio, 2025-04-02:
Regarding your first suggestion, the documentation would be a great place for explaining better how it works.

Regarding your second suggestion: there are at least three hardware revisions of the GC-01. So what works with your board doesn’t work with others.

closed 2025-04-03 by Gissio