Gissio/radpro #238: Fnirsi GC-01 2.1.1 update now no sound, vibration or LED.

santalant, 2025-04-09:
Installed 2.1.1 on Fnirsi GC-01 with APM32F103RBT6 with the install file not update version from 2.0rc5. Now no sound, vibration nor led with the settings for those turn on. When I power up it shows LED, vibrates and clicks just for a second after that its all back to none of it. It all worked great with 2.0rc5. Where can I get that file again? or the fix for this issue? Thanks in advance.

Gissio, 2025-04-10:
This seems to be a duplicate of #233 .

Can somebody else confirm? A video would be most useful.

Gissio, 2025-04-10:
Do you mean the Pulses options feet disabled? If that is the case, what happens if you re-enable them?

santalant, 2025-04-11:
Yes I meant pulses and it didn't matter when I cycled it on and off. I think I'll stick with 2.1 until the next iteration.

Gissio, 2025-04-11:
But were you able to re-enable them through the menu?

Gissio, 2025-04-12:
I don’t understand. Do you mean 2.1.1 works without issues now?

santalant, 2025-04-12:
Yes no issues after complete power removal including backup battery.Sent from my iPhone

Gissio, 2025-04-12:
Excellent. Closing issue.

closed 2025-04-12 by Gissio