Gissio/radpro #243: 2.1.1 FS-5000 openocd error (includes workaround)

byteborg, 2025-04-17:
FS-5000
radpro-flashtool-2.1.1 Win64
Programming with STLINK V2 clone `STLINK V2J17S4 (API v2) VID:PID 0483:3748`

The included openocd tries connecting using DAP transport (which is not available on cheap clones) and fails.

Workaround is using `lib\openocd\scripts\interface\stlink.cfg` from the previous release `radpro-flashtool-2.1` which defaults to HLA mode and everything is fine again.

Gissio, 2025-04-17:
Karsten: your report is doubly appreciated as you provide a solution.

In Rad Pro 2.1.1 I updated OpenOCD. What you report is strange. I tested this release by flashing an FS2011 counter with an ST-LINK V2 clone and it worked with no issues.

Could you please enable verbose mode in OpenOCD and share the log? Thanks in advance.

NuclearPhoenixx, 2025-04-17:
Funnily enough, I was able to flash the latest release of 2.1.1 just fine on my Win 11 machine. I too am using a cheap STLINK V2 clone. Maybe there was come kind of error message for me as well, I cannot remember, however, at the end the process concluded just fine and was successful without any apparent issues.

Gissio, 2025-04-18:
Byteborg: did you update your ST-LINK V2 to the latest ST firmware? I did—this could explain what we observe.

Anyway, I’ll specify hla_swd transport to the scripts in the next release.

Gissio, 2025-05-04:
Closing as there is no follow up.

closed 2025-05-04 by Gissio

byteborg, 2025-05-21:
Sry for not following up timely. Yes, the ST-LINK V2 was not upgraded. I upgraded it to the latest release and DAP mode worked flawlessly.

callimero, 2025-09-22:
Thanks that resolved my problem also!