Gissio/radpro #245: Exceptions on datalog export with radpro-tool

u-foka, 2025-04-30:
Hi,

I just tried to export my datalog and had two different exceptions when for some reason some of the log entries came out malformed causing radpro-tool.py to exit with an exception without even exporting partial data.

In my opinion exporting lines that can be parsed would still be better than loosing everything, so I've made some adjustments to the script (PR incoming :))

```
 :: ~ » radpro-tool --port /dev/cu.usbserial-313120 --download-datalog ~/Unclutter/radpro.csv
Syncing time...
Downloading data log...
Traceback (most recent call last):
  File "/Users/u-foka/Workspace/Development/Embedded/radpro/tools/radpro-tool.py", line 501, in <module>
    download_datalog(device,
    ~~~~~~~~~~~~~~~~^^^^^^^^
                     args.datalog_file,
                     ^^^^^^^^^^^^^^^^^^
                     args.datalog_datetime)
                     ^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/u-foka/Workspace/Development/Embedded/radpro/tools/radpro-tool.py", line 195, in download_datalog
    record_datetime = str(datetime.fromtimestamp(record_time))
                          ~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^
ValueError: year 555212 is out of range
 :: ~ » radpro-tool --port /dev/cu.usbserial-313120 --download-datalog ~/Unclutter/radpro.csv
Syncing time...
Downloading data log...
Traceback (most recent call last):
  File "/Users/u-foka/Workspace/Development/Embedded/radpro/tools/radpro-tool.py", line 501, in <module>
    download_datalog(device,
    ~~~~~~~~~~~~~~~~^^^^^^^^
                     args.datalog_file,
                     ^^^^^^^^^^^^^^^^^^
                     args.datalog_datetime)
                     ^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/u-foka/Workspace/Development/Embedded/radpro/tools/radpro-tool.py", line 193, in download_datalog
    pulse_count = int(values[1])
                      ~~~~~~^^^
IndexError: list index out of range
```

Gissio, 2025-04-30:
Instead of addressing the symptom, let's target the root cause. What device are you using?

u-foka, 2025-05-01:
Sure, I'm working on a repro, but I thought it would be still worth fixing the sciprt, question is if you wanna track both issues in the same issue, or do we open another one for the (potential) corruption?

Anyhow its a bosean fs-5000 with a j321 tube, arrived about a week ago, got radpro ~10 minutes later ;)

It was set to 10s data logging and the export i had the error with resulted (with the fixed script ignoring 6 entries)in a ~18k lines csv.

Been looking at the data as it seemed to be that in my house the average background radiation was ~50 nS/h higher than elsewhere.. Ended up putting it in different drawers in my home office finding my granpas old russian compass, when putting the device directly on top it had readings fluctuating between 5-10 μS/h... Was vondering if this is a sufficiently high level to cause bitflips in the flash, but now I realized that it probably wasnt the flash, as every download attempt resulted in different errors...

Afterwards i've reset the data and set it to 10 minute intervals for logging, and never saw the problem again.

Now I've set it to 1s, will see if the problem triggers due to the ammount of data in the log. It took quite some time to push through that 18k lines the serial interface :)

Not sure if it is of any use, but here is the final successfull export and the resulting file for reference.
```
 :: ~ » radpro-tool --port /dev/cu.usbserial-313120 --download-datalog ~/Unclutter/radpro.csv
Syncing time...
Downloading data log...
Warning: invalid data log format in line 11029 is skipped:
Warning: invalid data log format in line 11372 is skipped: 1745944186914
Warning: invalid data log format in line 15649 is skipped: 1745986967,1077,109703
Warning: invalid data log format in line 16314 is skipped: 1745992
Warning: invalid data log format in line 17276 is skipped: 1746003241,1746003251,116771
Warning: invalid data log format in line 17597 is skipped: 1746006461,11781,117814
Logging live...
```

[radpro.csv](https://github.com/user-attachments/files/20004032/_radpro.csv)

Gissio, 2025-05-01:
The issue seems to stem from the download process rather than the logging itself. We need to determine whether electrical noise is interfering with the serial port or if the problem is software-related.

Could you try downloading the log while the display is in sleep mode and check if the error rate decreases?

u-foka, 2025-05-01:
Just tried with the so far accumulated 3000 entry long log, no (visible) issue so far. But if its some electrical issue it means any byte can be corrupt, just not all would cause parsing errors.

Does this thing have an uart to usb chip? I see no other way how interference could cause this, the usb serial protocol should catch such problems I believe. Gonna try to find a schematic and check, and will continue trying to reproduce, and if I could then will retest with screen off.

Gissio, 2025-05-02:
Once you have a sufficiently large datalog, please try the following tests:

* Download the datalog 3 times without erasing it, keeping the display on and using the device's keys to navigate the menus during the download. Record and report the total number of radpro-tool errors encountered.
* Download the datalog 3 more times, this time keeping the screen off and without navigating the menus. Record and share the total number of radpro-tool errors encountered.

If the issue is hardware-related, the number of errors from both tests should be similar. If the issue is software-related, the number of errors should differ significantly.

Gissio, 2025-05-04:
Thank you for sharing the log! After reviewing it, I've confirmed that the issue is software-related, as the log shows only deletions with no insertions or substitutions.

I'll need some time to analyze this further.

u-foka, 2025-05-04:
Nice, thats good news 😆 Meanwhile I tried to change baud, and flash my compiled firmware but somehow it errored out mid flash, and I couldn't recover from it yet, so we'll see if I can test the fix for you anytime soon 😅

Gissio, 2025-05-09:
I've just tested the serial port interface, and Rad Pro seems to work allright.

Can some other Bosean FS-5000 users confirm this problem?

Gissio, 2025-07-01:
u-foka: It seems you have an EMI problem. Try downloading the logs using a battery-powered laptop. Does this resolve the problem?

closed 2025-07-01 by Gissio