Gissio/radpro #272: FNIRSI GC-01 (2.2beta5):  pulse indication per sound is gone while display sleep mode

RoboSchmied, 2025-07-31:
When 2 indicators (LED + sound) are enabled:
The sound pulse indication is switched off, when the display goes in sleep mode.
LED pulse indication is working in sleep mode.

When all 4 indicators are enabled, all are working.

When 3 indicators (LED, sound, vibration) are enabled, only sound is not working while display is in sleep mode.

OK, the bug is:  **sound indication is only working (in display sleep mode) when `Display flash` is enabled too**.

Gissio, 2025-07-31:
Thank you for your report.

Im having trouble understanding the issue. You mentioned  that clicks occur when the display is off in these scenarios:

* Sound and LED enabled; vibration and display flashes disabled.
* Sound, LED, vibration, and display flashes enabled.

However, you noted no clicks occur when the display is off in these scenarios:
* Sound, LED, and vibration enabled; display flashes disabled.

Could you confirm this is accurate?

As I don’t currently have a GC-01, I’m relying on your reports to diagnose the issue.

Gissio, 2025-08-01:
The issue stemmed from an optimization incompatible with the GC-01!

Could you test this: https://github.com/Gissio/radpro/actions/runs/16666922314/artifacts/3664493472

Gissio, 2025-08-01:
The issue stemmed from an optimization incompatible with the GC-01!

Could you test this? ~~https://github.com/Gissio/radpro/actions/runs/16666922314/artifacts/3664493472~~
https://github.com/Gissio/radpro/actions/runs/16667330084/artifacts/3664611650

closed 2025-08-02 by Gissio