Gissio/radpro #274: Show Batterie loading for FS-600 if turned off

SunboX, 2025-08-05:
If the FS-600 is turned off and is charging it's batteries, please let the LED slowly blink. If it is fully charged, the LED should light up constantly as long as USB connector is plugged in.

If Batteries are not present or set to "Alkaline", the LED should turn off if Device is off and connected to USB.

Gissio, 2025-08-05:
In the upcoming release, you can check the charge status of a powered-off device by pressing any key, similar to a cellphone. Hang tight.

Gissio, 2025-08-05:
In UI design, using a single indicator for multiple purposes can create confusion.

Gissio, 2025-08-05:
In the upcoming release, plugging in a device shows the charging screen for five seconds, suggesting you can press any key to see it again.

I won’t implement what you suggest.

Gissio, 2025-08-07:
Converting to discussion (#32).