Gissio/radpro #282: 3.0rc1 firmware not found

MrJanSv, 2025-08-27:
The installation of version 3.0rc1 cannot be started, after entering the language cs it says firmware image not found

Gissio, 2025-08-27:
What device do you have?

MrJanSv, 2025-08-27:
FS5000

Gissio, 2025-08-27:
There’s a typo in install.bat/install.sh. For now, Can you edit rhe install file you are using with a text editor and change `fs-5000` to `fs5000`?

Gissio, 2025-08-27:
That is very helpful. Thanks a lot.

MrJanSv, 2025-08-27:
so I uploaded the older version and then the new one again and it works now, thank you

MrJanSv, 2025-08-27:
so no, after turning it off and on again it froze again :/

Gissio, 2025-08-27:
What happens when you remove the battery and reinsert it?

Hantervit, 2025-08-28:
Hello.

https://github.com/user-attachments/assets/7e31ce1b-a0d3-4f87-85da-f4667b756b50

Hantervit, 2025-08-28:
with batery

https://github.com/user-attachments/assets/d613432c-9910-4fd4-8330-4e69e8b39396

Gissio, 2025-08-28:
So it stops working after the first beep. That's great news, as it means that the RTC clock is not starting up correctly. I'll keep you updated.

Gissio, 2025-08-28:
Can you try again?

[radpro-bosean-fs5000-en-3.0rc1-test3.zip](https://github.com/user-attachments/files/22018071/radpro-bosean-fs5000-en-3.0rc1-test3.zip)

Gissio, 2025-08-28:
Great news. So I guess we are ready to release 3.0!

closed 2025-08-28 by Gissio

Hantervit, 2025-08-28:
I can check the new version only in the evening. I will write about the result.

Hantervit, 2025-08-28:
> Can you try again?
> 
> [radpro-bosean-fs5000-en-3.0rc1-test3.zip](https://github.com/user-attachments/files/22018071/radpro-bosean-fs5000-en-3.0rc1-test3.zip)

This version works! Thank you!

Ilja-code-art, 2025-08-28:
firmware checksum failure....... 

> > Can you try again?
> > [radpro-bosean-fs5000-en-3.0rc1-test3.zip](https://github.com/user-attachments/files/22018071/radpro-bosean-fs5000-en-3.0rc1-test3.zip)
> 
> This version works! Thank you!

And no: "firmware checksum failure"???

Ilja-code-art, 2025-08-30:
> > firmware checksum failure.......
> > > > Can you try again?
> > > > [radpro-bosean-fs5000-en-3.0rc1-test3.zip](https://github.com/user-attachments/files/22018071/radpro-bosean-fs5000-en-3.0rc1-test3.zip)
> > > 
> > > 
> > > This version works! Thank you!
> > 
> > 
> > And no: "firmware checksum failure"???
> 
> "firmware checksum failure" This is not a mistake. Just a test firmware file. And Gissio did not sign it with his signature, so as not to spend time on this. This does not affect work in any way, you can safely ignore this message.

Thanks a lot.