Gissio/radpro #283: FNIRSI GC-01 (CH32): Firmware 3.0 rc 1 the time.

peshka110, 2025-08-28:
FNIRSI GC-01 (CH32): Firmware 3.0 rc 1. If I set the time, it appears near the battery. After turning it off and on, the time near the battery disappears. And in the settings it resets to factory settings, you have to select the date and time again. And I don’t understand why we need a date when it’s not on the display. For a log? Maybe you’re young and don’t need a date. Under real conditions and exposure to radiation, not only the time and date will fly out of your head, but also what city you’re in. The rest of the settings seem good. Thanks for the new firmware, I was really waiting for it.

peshka110, 2025-08-28:
I'll install the old firmware, configure it and turn it off. If the time doesn't change, then the СR 1220 battery is working.

Gissio, 2025-08-28:
Use a voltmeter!

peshka110, 2025-08-28:
Yes, on firmware 2.2b5 the date and time also got lost, which means my sr 1220 died. Sorry. I'll open it now and take measurements. Thank you for the tip.

Gissio, 2025-08-28:
Let us know if it worked, and if you like the new history view that loads the data log in startup.

peshka110, 2025-08-28:
Yes GR1220 charge 0 volt.

peshka110, 2025-08-28:
The story is good. But with the day theme of the display everything is fine, but with the twilight and night theme mode the grille is weakly visible. That's how it seemed. Thanks again.