Gissio/radpro #300: BOSEAN FS-600 update script not working on macOS (v3.0.1)

SunboX, 2025-09-05:
Console output:

```sh
bosean-fs600 % ./install.sh 
Available language codes:
Enter language code for Rad Pro installation: de
../bosean-fs-600/firmware/radpro-bosean-fs600-de-
Firmware image not found.
```

Folder:

[image]

SunboX, 2025-09-05:
I'm using a MacBook Pro with Apple M3 Max silicon

Gissio, 2025-09-05:
There's an error in the install script. Change "bosean-fs-600" to "bosean-fs600" and it should work.

Will be fixed in the next release.

SunboX, 2025-09-07:
Thanks a lot! I will wait for the next release. Don't want to screw open the device more often, have sadly not much spare time. 😔

Gissio, 2025-09-07:
If you’re aleeady on 3.0, you should be able to update with the new [web installer](https://gissio.github.io/radpro-installer/). Have you tried it?

SunboX, 2025-09-09:
I installed Release 3.0.2 using the `install.sh` which worked fine. But the Device does not turn on afterwards.

Log:
```
./install.sh
Available language codes: bg cs da de el en es fi fr hr hu it ja ko nl no pl pt ro ru sk sl sv tr uk vi zh_CN
Enter language code for Rad Pro installation: de
../bosean-fs600/firmware/radpro-bosean-fs600-de-3.0.2.bin
Backing up old firmware image...
xPack Open On-Chip Debugger 0.12.0+dev-01850-geb6f2745b-dirty (2025-02-07-12:10)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
DEPRECATED: OpenOCD support for ST-Link HLA transport will be dropped soon!
Consider updating your ST-Link firmware to a version >= V2J24 (2015)
Info : auto-selecting first available session transport "hla_swd". To override use 'transport select <transport>'.
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
Info : clock speed 2000 kHz
Info : STLINK V2J37S7 (API v2) VID:PID 0483:3748
Info : Target voltage: 3.243501
Info : [stm32g0x.cpu] Cortex-M0+ r0p1 processor detected
Info : [stm32g0x.cpu] target has 4 breakpoints, 2 watchpoints
Info : [stm32g0x.cpu] Examination succeed
Info : [stm32g0x.cpu] starting gdb server on 3333
Info : Listening on port 3333 for gdb connections
Info : Unable to match requested speed 2000 kHz, using 1800 kHz
Info : Unable to match requested speed 2000 kHz, using 1800 kHz
[stm32g0x.cpu] halted due to debug-request, current mode: Thread 
xPSR: 0xf1000000 pc: 0x080006fc msp: 0x20009000
Flashing new firmware image...
xPack Open On-Chip Debugger 0.12.0+dev-01850-geb6f2745b-dirty (2025-02-07-12:10)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
DEPRECATED: OpenOCD support for ST-Link HLA transport will be dropped soon!
Consider updating your ST-Link firmware to a version >= V2J24 (2015)
Info : auto-selecting first available session transport "hla_swd". To override use 'transport select <transport>'.
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
Info : clock speed 2000 kHz
Info : STLINK V2J37S7 (API v2) VID:PID 0483:3748
Info : Target voltage: 3.241910
Info : [stm32g0x.cpu] Cortex-M0+ r0p1 processor detected
Info : [stm32g0x.cpu] target has 4 breakpoints, 2 watchpoints
Info : [stm32g0x.cpu] Examination succeed
Info : [stm32g0x.cpu] starting gdb server on 3333
Info : Listening on port 3333 for gdb connections
Info : Unable to match requested speed 2000 kHz, using 1800 kHz
Info : Unable to match requested speed 2000 kHz, using 1800 kHz
[stm32g0x.cpu] halted due to debug-request, current mode: Thread 
xPSR: 0xf1000000 pc: 0x080006fc msp: 0x20009000
Info : device idcode = 0x20006460 (STM32G07/G08xx - Rev B : 0x2000)
Info : RDP level 0 (0xAA)
Info : flash size = 128 KiB
Info : flash mode : single-bank
Info : Unable to match requested speed 2000 kHz, using 1800 kHz
Info : Unable to match requested speed 2000 kHz, using 1800 kHz
[stm32g0x.cpu] halted due to debug-request, current mode: Thread 
xPSR: 0xf1000000 pc: 0xfffffffe msp: 0xfffffffc
** Programming Started **
** Programming Finished **
Info : Unable to match requested speed 2000 kHz, using 1800 kHz
Info : Unable to match requested speed 2000 kHz, using 1800 kHz
```

SunboX, 2025-09-09:
3.0.1 works if I fix the installer naming

Gissio, 2025-09-09:
Can you try again with this file:

[radpro-bosean-fs600-en-3.0.2.bin.zip](https://github.com/user-attachments/files/22238410/radpro-bosean-fs600-en-3.0.2.bin.zip)

SunboX, 2025-09-10:
Will try this evening (round about 11 hours)

Gissio, 2025-09-10:
Take it easy. Thanks a lot!

Gissio, 2026-02-08:
Read the second message of #357.

But remember: it is a beta. If it doesn’t work, you have to recover with an ST-LINK.

Gissio, 2026-02-24:
Does it work with 3.1beta2?